polyend/TrackerBetaTesting #151: [1.1.18] MIDI-in stuck notes

djkurzman, 2020-10-06:
Reporting again after more testing on 1.1.18

Since 1.1.15, I have had issues with MIDI notes becoming stuck when played in quick succession with some brief overlap. I can replicate the issue with any 3 chords played from midi input.

--This does not happen with the Tracker keys.
--Notes play normally when in Rec+Play mode.
--Notes play normally in the Sample Playback screen.

Tested with MIDI DIN (with and without USB host) using multiple MIDI controllers and direct MIDI USB. Changed all cables, no dust in midi in jack.

Notes must be played in quick succession with light overlap to reproduce. It can happen with any 3 or more notes played. This happens in any scenario except those listed above.

[polytrackermidi01.zip](https://github.com/polyend/TrackerIssues/files/5330676/polytrackermidi01.zip)

In the attached MP3 I use the styles described below starting in the Sample Playback screen to demonstrate how the notes should sound. I then play the notes again from the Instrument Performance screen to show the issue.
Each demonstration starts with a simultaneous chord press (C-E-G), then the same notes one after another, then in quick succession while holding the G key. You can hear in the second run that G only plays briefly, but E continues to play until I release the G key. I have found that if any key is pressed while E is stuck, E will continue to play (even if pressing the E Key again). It will only stop playing if all other keys are released.

USB MIDI captured via MIDIOX

All keys simultaneous press -- All notes play as expected (Pattern Screen-Rec off)
 TIMESTAMP IN PORT STATUS DATA1 DATA2 CHAN NOTE EVENT               
 0004DA0E  KEY  2     90    3C    64    1  C  4 Note On               
 0004DA0F  KEY  2     90    40    64    1  E  4 Note On               
 0004DA0F  KEY  2     90    43    64    1  G  4 Note On               
 0004DF8A  KEY  2     80    3C    40    1  C  4 Note Off              
 0004DF8A  KEY  2     80    40    40    1  E  4 Note Off              
 0004DF8B  KEY  2     80    43    40    1  G  4 Note Off              

One key after another, no note overlap -- All notes play as expected (Pattern Screen-Rec off)
 TIMESTAMP IN PORT STATUS DATA1 DATA2 CHAN NOTE EVENT               
 0004E486  KEY  2     90    3C    64    1  C  4 Note On               
 0004E554  KEY  2     80    3C    40    1  C  4 Note Off              
 0004E645  KEY  2     90    40    64    1  E  4 Note On               
 0004E759  KEY  2     80    40    40    1  E  4 Note Off              
 0004E812  KEY  2     90    43    64    1  G  4 Note On               
 0004E91F  KEY  2     80    43    40    1  G  4 Note Off              

Keys played in quick succession with brief overlap -- Stuck notes occur (Pattern Screen-Rec off)
 TIMESTAMP IN PORT STATUS DATA1 DATA2 CHAN NOTE EVENT               
 0004ED23  KEY  2     90    3C    64    1  C  4 Note On               
 0004ED6A  KEY  2     90    40    64    1  E  4 Note On               
 0004ED8E  KEY  2     80    3C    40    1  C  4 Note Off              
 0004ED8E  KEY  2     90    43    64    1  G  4 Note On <--G note only plays briefly while I am holding the key down
 0004EDBF  KEY  2     80    40    40    1  E  4 Note Off <--E note continues to play on the tracker even though E key is released
 0004F806  KEY  2     80    43    40    1  G  4 Note Off <--Once the G key is released, E note stops playing
 

Keys played in quick succession with brief overlap -- All notes play as expected (Pattern Screen-Rec+Play mode)
 TIMESTAMP IN PORT STATUS DATA1 DATA2 CHAN NOTE EVENT               
 00142A32  KEY  2     90    3C    64    1  C  4 Note On               
 00142A64  KEY  2     90    40    64    1  E  4 Note On               
 00142A95  KEY  2     80    3C    40    1  C  4 Note Off              
 00142AB9  KEY  2     90    43    64    1  G  4 Note On               
 00142ACE  KEY  2     80    40    40    1  E  4 Note Off              
 00142D1B  KEY  2     80    43    40    1  G  4 Note Off              
 00142EE8  KEY  2     90    3C    64    1  C  4 Note On               
 00142F27  KEY  2     90    40    64    1  E  4 Note On               
 00142F59  KEY  2     80    3C    40    1  C  4 Note Off              
 00142F8C  KEY  2     80    40    40    1  E  4 Note Off              
 00142F8C  KEY  2     90    43    64    1  G  4 Note On               
 001431FC  KEY  2     80    43    40    1  G  4 Note Off

Jacknametrouble, 2020-10-06:
Hey, just checked again over MIDI DIN and cannot reproduce anyhow, been playing chords up to 8 notes, strums, overlapping notes etc. for a longer amount of time but no success in reproducing. Any chance you could catch this on a video?

Jacknametrouble, 2020-10-06:
I think I finally got this! The notes are not getting stucked at all, they're being cut and this happens only in the Sample Editor in my case.

djkurzman, 2020-10-06:
Finally got a better video recording of the issue. Sorry if the volume is a bit low. I know I mentioned previously that the keys had to be played in quick succession. Really it only requires two notes to overlap, and then for any other keys to be played. Hopefully the video is helpful.
[Vid 20201006 173255-1.zip](https://github.com/polyend/TrackerIssues/files/5337042/Vid.20201006.173255-1.zip)

djkurzman, 2020-10-07:
> I think I finally got this! The notes are not getting stuck at all, they're being cut and this happens mostly in the Sample Editor section and is because there are no envelope values attached to the instrument.

I had not done any testing in the Sample Editor screen for the reason you mentioned. I did test from all other screens and was able to produce the stuck notes, except in the Sample Playback screen when everything plays normally. Notes will get stuck in the Pattern screen unless in Rec+Play mode. I tested with Rec and then Play mode separately and the issue was still present.

I've tried reverting back to older firmware to try to resolve this, but it continued even on 1.1.0. This issue was definitely not occurring until I upgraded to 1.1.15. At this rate I am willing to try any troubleshooting step.

djkurzman, 2020-10-07:
Adding a copy of my SD contents except for samples and projects. The workspace project is included since that is what I last used to demonstrate the issue.
[tracker backup.zip](https://github.com/polyend/TrackerIssues/files/5343090/tracker.backup.zip)

djkurzman, 2020-11-12:
Update - I tried rolling all the way back to 1.0.0 and nothing changed. Stuck notes continued happening somehow. Upgraded back to 1.2.2 again and found the same.

closed 2021-01-19 by ambv