polyend/TrackerBetaTesting #157: [1.1.18] Sample editor previews slowly get worse and then the Tracker's audio stops entirely

ArtFluids, 2020-10-06:
So I was testing out the new sample conversion, it works great for the most part. The samples sound good/right etc.
However as I was previewing samples they were slowly getting more and more fuzzy/weird and eventually the Tracker stopped playing audio entirely. I reset the unit and things were fine.

I was previewing the samples I mentioned in a previous ticket here: https://github.com/polyend/TrackerIssues/issues/86
Try loading lots of them into a project and paying attention to the audio quality of the previews. Eventually the previews sound nothing like the original sample.

There's probably still some strange issues with the samples being 8-bit and throwing off your memory space or something. Like I said, it was the preview that was causing problems, when the samples are loaded/converted then everything is ok, presumably because there's no more 8-bit data in play.

wjakobczyk, 2020-10-06:
Hi, in the last paragraph you stress that the issue is in sample preview (in Sample Loader I assume), but earlier you state "Try loading lots of them into a project" which is a bit puzzling. Could you clarify?

piotr3500, 2020-10-06:
@ArtFluids can you show us the issue on the video?

ArtFluids, 2020-10-06:
> Hi, in the last paragraph you stress that the issue is in sample preview (in Sample Loader I assume), but earlier you state "Try loading lots of them into a project" which is a bit puzzling. Could you clarify?

Yes, in the Sample Loader. I had moved several sounds over to the project and noticed it was not letting me preview the project samples anymore. I pressed left back into the folder of samples and started trying to listen to those and they sounded fuzzy and weird, and after a while they stopped playing entirely. So I went back to my pattern and even that didn't make sound anymore. The samples in the project didn't get corrupted because when I reset the Tracker then everything was ok.

> @ArtFluids can you show us the issue on the video?

Yeah let me see if I can make a video soon.

ArtFluids, 2020-10-06:
@piotr3500 
@wjakobczyk 
Here's a video
https://youtu.be/7MSe0E56rwI

ArtFluids, 2020-10-06:
The only difference with that video is that the samples sounded ok the whole time, but the first time this happened to me the samples started to sound weirder and weirder before going silent

piotr3500, 2020-10-06:
Thanks for the video. It helps a lot.
Did it happen on 1.1.17?

ArtFluids, 2020-10-06:
I did not try this extensively with 1.1.17, because if you look at my previous issue it wasn't loading the 8-bit samples properly at all so I never spent so much time with the Sample Loader when I was using 1.1.17

ArtFluids, 2020-10-06:
I only know now because it actually does load the 8-bit samples properly, I can use them in my project now and they sound nice

keith-macleod, 2020-10-07:
I have encountered this or something very similar with sample preview. It doesn't matter how many samples I have in the project, it can happen with no samples loaded.
Seems like it happens when I'm previewing a bunch of samples quickly, maybe related to switching the sample while the sample is still playing, but there is at least one other factor, because it won't always happen when I do so.
Either way, all audio stops playing, and I need to restart.

stevesomeone, 2020-10-07:
> I have encountered this or something very similar with sample preview. It doesn't matter how many samples I have in the project, it can happen with no samples loaded.
> Seems like it happens when I'm [...]

I have also come across this behaviour.

closed 2020-10-07 by andrzejmagierski2