polyend/TrackerBetaTesting #158: 1.1.15->1.1.18 MIDI input not recorded

fotografpaul, 2020-10-06:
## Hi, sorry you had trouble!

### Steps to reproduce
Select a midi instrument and use an external controller to play/record it.

It plays without any issues, but when you record no notes are recorded. 

[image]

Previous version 1.1.14 and back it records but does not trigger the actual MIDI Instrument. 

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Every time

* Does this bug happen in previous firmware versions?
From 1.1.15 to 1.1.18 its present. 

* Does this bug happen in every project?
Yes

piotr3500, 2020-10-06:
Cannot reproduce. Perhaps you are using a wrong adapter?

fotografpaul, 2020-10-06:
It's been working _(well it's silent when you record since it won't trigger the external midi instrument, only record the notes from the external controller)_  since the latest stable firmware and up to 1.1.14 after that, any external input won't be recorded, but playing works. Adapters are original Polyend and have been working fine since day one. :-) 

In other words, the MIDI issue is now reversed, before the notes would be recorded (when in record mode) but you would not hear anything since the tracker would not trigger the external MIDI instrument. Now when in record mode the Tracker does trigger the Instrument but does not record any notes.

piotr3500, 2020-10-06:
Got it. Thanks.

closed 2020-10-06 by shuler323