polyend/TrackerBetaTesting #162: [1.1.18] Volume effect command, inconsistent with dB values / bricked songs

MegaTurrican, 2020-10-06:
Now that the instrument volume is represented in dBs, all the songs that were programmed in the previous firmwares and specifically with the volume command are inconsistent; the volume is much higher than expected. It is also difficult to calculate the volume entry with the command, since on the command you have 0 to 100 and in the instrument volume you have dBs.

Jacknametrouble, 2020-10-06:
We've introduced a mechanism which is "recounting" the volumes and checked that it's working properly on some of our projects with positive effect. Can you provide any particular projects where this is noticeable with the before/after reference audio example, please?

MegaTurrican, 2020-10-06:
It occurs in granular & wavetable instruments. I will send you later the project files :)

piotr3500, 2020-10-06:
The problem was not on FX but Instrument Volume. This will be fixed in 1.1.19.

closed 2020-10-06 by piotr3500

MegaTurrican, 2020-10-06:
Great :D Keep up the excellent work guys!

MegaTurrican, 2020-10-07:
@piotr3500, @Jacknametrouble  The issue also occurs on 1.1.19 - I just loaded the same song and the same thing happens. The dB instrument volume scales are not relative to the V FX scales. Same instrument sounds much louder in 1.1.18 & 1.1.19 when programming a V FX automation (from V01 to V40). In the firmwares before (1.1.17 and bellow) this FX automation had a progressive and quiet resolution, where now its very loud and more abrupt scaled.

piotr3500, 2020-10-07:
You are right. We'll fix it.