polyend/TrackerBetaTesting #165: [V1.1.18] Midi Playback scrambled in Sample Recording

disastrou, 2020-10-06:
## Hi, sorry you had trouble!

### Steps to reproduce

External midi playback works well ok the sample recording screen, but when you actually hit record to attempt to resample external midi, the timing appears to scramble.  See attached video:

https://www.dropbox.com/s/wwlqwmkixeqd8zp/IMG_3044.MOV?dl=0

This is universal to all projects and with any midi device I’ve tried it with.

Jacknametrouble, 2020-10-08:
Seems that the recorded audio itself is alright. Isn't it just the thing that both the MIDI controlled instrument is audible at the same time when the currently recorded audio is heard too and there's a bit of difference between them so it sounds like it's scrambled/doubled up?

disastrou, 2020-10-08:
I don’t *think* that’s it.  I just tried again and unplugged the line in after recording to check:

https://www.dropbox.com/s/h59bkpq0l82ri8r/IMG_3063.MOV?dl=0

Jacknametrouble, 2020-10-08:
Indeed, sounds like it almost doubled up in speed. Thank you for the video.

bambrose, 2020-10-11:
Yeah I have the same. And Yes, I agree it seems to be doubling in speed. Have a video if that helps. but I think covered off by others already

closed 2020-11-12 by shuler323