polyend/TrackerBetaTesting #167: More modulation control

keith-macleod, 2020-10-06:
I don't know what kind of sound design improvements you have on your roadmap, but there are a few relatively simple modulation enhancements that would make the wavetable synthesis engine much more usable, as well as being just generally useful:
- Envelope polarity
- Envelope retrigger
- LFO Unipolar
- LFO phase
- Unsynced LFO (This is also a feature request elsewhere)

BTW, thanks for the great work you've already done. I'm very glad that you're continuing development and making point releases publicly available.

keith-macleod, 2020-10-06:
Also, not strictly modulation, but key tracking for the filter would also be helpful for sound design.

sebsonf, 2020-10-12:
I noticed that the lfo is reset with each voice / note. It would be nice to have the option for a global / per instrument free running lfo (e.g. for panning).

closed 2021-12-02 by Jacknametrouble