polyend/TrackerBetaTesting #171: 1.1.19 No information recorded in Live Rec mode

stevesomeone, 2020-10-07:
When Live Rec mode is active, no information is recorded on the track. The track is not displaying anything as being recorded, however the track is playing the audio corresponding to what you have recorded. 

Often the information will be recorded on another track entirely (not the track in focus).

When this occurs, it is often the case that I cannot step out of Live Rec mode into Step mode.

Will try to capture it on Video and update here with a link.

* How often does this bug happen? Behaviour is not consistent, but can be reproduced.

* Does this bug happen in previous firmware versions? Yes, this has happened on all previous firmwares.

* Does this bug happen in every project? Yes, it happens regardless of project   ##opened.

piotr3500, 2020-10-11:
Please try 1.1.22 and let us know if it still happens. It shouldn't... 
Cheers.

stevesomeone, 2020-10-11:
Hi, unfortunately it is still a problem.

I have included 2 video links displaying the lack of recording and random recording onto different tracks as well as the inability to step back to step rec mode.

Midi clock and transport over jack.

https://www.dropbox.com/s/rv9q54wxrabivuf/Live%20rec%20bug%201.mp4?dl=0
https://www.dropbox.com/s/xpngjy0py0outna/Live%20rec%20bug%202.mp4?dl=0

closed 2020-10-16 by piotr3500