polyend/TrackerBetaTesting #172: 1.1.19 Sample playback head display

stevesomeone, 2020-10-07:
Focus on a track which has been muted.
Go to Sample Loader screen.
Preview sample with either pads or preview button.
The red playhead indicator does not accurately reflect retriggering of the sample.
Audio playback of the sample being retriggered is correct.
Sequencer is playing. No information on track to interfere with playback.

Video Link here. [https://www.dropbox.com/s/gsdbmhzxj7gujix/sample%20playback%20indicator.mp4?dl=0](url)

* How often does this bug happen? All the time.

* Does this bug happen in previous firmware versions? Yes, all versions so far

* Does this bug happen in every project? Yes, every project.

piotr3500, 2020-10-07:
If you have this instrument somewhere in the pattern, then it overtakes the preview.
Is this the case?

stevesomeone, 2020-10-07:
> 
> 
> If you have this instrument somewhere in the pattern, then it overtakes the preview.
> Is this the case?

Sometimes that is the case, yes. Even when I am focused on a muted track that does not contain that instrument, the sample playback head will behave this way. It seems to be following some sort of trigger pattern, possibly from another track.

I have deliberately focused on a track (active and playing) and opened the sample playback to preview the same sample and there is no problem: it will behave polyphonically without choking itself off when a step in the pattern plays. So I can't work out what is happening and why?

The sample playback indicator will sometimes freeze completely.

stevesomeone, 2020-10-07:
I realise that the Tracker was probably not designed to be used this way, but it is curious that you can indeed do that, but not consistently and not with any sample.

Jacknametrouble, 2020-10-08:
The played sample head display is just a visual hint for an orientation. It's not supposed to give full feedback when playing chords etc (more than one head).

closed 2020-10-08 by Jacknametrouble