polyend/TrackerBetaTesting #175: Easy features to implement

Sft-mlk, 2020-10-07:
Currently at _v1.1.0_ :

   -**A direct access to _'Send-Fx's Parameters'_ from the _'Instruments Sends'_ window.** _(As an alternative to _'Master'_ window access)_ _[Shift+Reverb / pushing it twice]_

   -**Select the entire track from its bottom step too.** [For example: Shift+down] 

   -**Faster way to scroll for granular parameters, specially _'Length'_.** _[(Shift+Knob]_

   -**The _'Volume Step FX'_ as a percentage of the original _'Instrument Volume'_, and not a free/master volume just for that step.** _(If you use it, you removes the ability to adjust the volume of an instrument from a single point/knob)_

   -**_'Micro-tune'_ exists as a FX, but not _'tune'_.** 

   -**Rename/remove/organize SD card files from the Tracker itself.**

jaap3, 2020-10-07:
Those are some good suggestions.

I don't have anything to do with Polyend, but I assume they'd like a new issue for each individual feature request. That way it's easier to track which feature request is implemented and which one isn't (yet). It also allows for discussion of each feature request in individual threads.

Also, don't assume features are easy to implement ;)

Sft-mlk, 2020-10-07:
> Those are some good suggestions.
>
> I don't have anything to do with Polyend, but I assume they'd like a new issue for each individual feature request. That way it's easier to track which feature [...]

I was afraid of saturate this page with request, but yes, make sense.
Yeah... 'Easy' word was just a word 👍

bambrose, 2020-10-08:
Tune as an FX is excellent.
Shift + Knob for fast scroll through values already exists elsewhere, but is another excellent suggestion.

closed 2021-12-02 by Jacknametrouble