polyend/TrackerBetaTesting #178: v1.1.19 Big bug related to instrument volume

Sft-mlk, 2020-10-07:
I just update to this version and the volume of some samples got crazy _(very **high levels**)_ especially on steps where I use _'Volume Fx **(V)**'_ and my intention was, paradoxically, turn down the volume of those steps.

I said 'especially' on those steps because I notice all instruments turn down their volumes in comparison to the same project with the v1.1.0 firmware. There was no relation between instrument volume and step Fx volume, I mean, I couldn't fix it turning down every volume of every instruments (traditional mixing technique) and reprogramming again the step Fx...

I hope I have explained myself correctly.

piotr3500, 2020-10-07:
Duplicate of #162

closed 2020-10-07 by piotr3500

Sft-mlk, 2020-10-08:
Oh, nice. It's the same bug for v1.1.18 and v1.1.19. I'm new reporting 'things about things', sorry! 👍