polyend/TrackerBetaTesting #180: Merge for Copy/Paste

djkurzman, 2020-10-07:
I think it would be nice to have a track merging option in the pattern view. When selecting multiple tracks, there would be options to merge all existing notes over to the first selected track with or without overwriting existing notes. This could be used to condense tracks without having to render them to an audio file.

For example, merging tracks 1 and 2 would result in the following
Track 1 | Track 2
------------ | -------------
C5 01 ---- ---- | G3 20 S 11 ----
C5 01 L 50 ---- | -- -- ---- ----
-- -- ---- ---- | G4 22 S 10 ----
-- -- ---- ---- | -- -- ---- ----
C5 01 ---- ---- | G3 20 S 11 ----
-- -- ---- ---- | G3 20 L 30 ----
C5 01 L 50 ---- | G4 22 S 10 ----
-- -- ---- ---- | -- -- ---- ----

Merge with overwrite
Track 1 | Track 2
------------ | -------------
G3 20 S 11 ---- | -- -- ---- ----
C5 01 L 50 ---- | -- -- ---- ----
G4 22 S 10 ---- | -- -- ---- ----
-- -- ---- ---- | -- -- ---- ----
G3 20 S 11 ---- | -- -- ---- ----
G3 20 L 30 ---- | -- -- ---- ----
G4 22 S 10 ---- | -- -- ---- ----
-- -- ---- ---- | -- -- ---- ----

Merge without overwrite
Track 1 | Track 2
------------ | -------------
C5 01 ---- ---- | -- -- ---- ----
C5 01 L 50 ---- | -- -- ---- ----
G4 22 S 10 ---- | -- -- ---- ----
-- -- ---- ---- | -- -- ---- ----
C5 01 ---- ---- | -- -- ---- ----
G3 20 L 30 ---- | -- -- ---- ----
C5 01 L 50 ---- | -- -- ---- ----
-- -- ---- ---- | -- -- ---- ----

HakkahFR, 2020-10-08:
Can be a real cool feature !

bambrose, 2020-10-08:
This would be a MASSIVE feature enhancement in my opinion.

PEtajaja, 2020-11-10:
I was just about to write a request for this! I was thinking that the track that you're pasting to could always take priority in regards to overlapping notes so that there's no need for paste with / without overwrite. It'd just always be without overwrite.

closed 2021-12-02 by Jacknametrouble