polyend/TrackerBetaTesting #183: 1.1.20 Change all instruments on a track with the jog wheel.

stevesomeone, 2020-10-08:
Scenario 1.
step record mode - select a step - press instrument - you can change the instrument via the pads or jog wheel.

Scenario 2.
step record mode - select all steps on track - press instrument - you can change the instrument via pads only. (cannot be changed with the jog wheel.

* How often does this bug happen? All the time

* Does this bug happen in previous firmware versions? All versions

* Does this bug happen in every project? Any project

piotr3500, 2020-10-08:
Can you show us the video?

stevesomeone, 2020-10-08:
Here is a video link. 
https://www.dropbox.com/s/6vcg4svlokv273m/select%20instrument%20jog%20wheel..mp4?dl=0

piotr3500, 2020-10-09:
@stevesomeone you tried all methods apart from the one that does the job :)
Go into Edit (red) mode. Select the steps. Press (not hold) Instrument button. Rotate the jog wheel.

closed 2020-10-09 by piotr3500

stevesomeone, 2020-10-09:
> 
> 
> @stevesomeone you tried all methods apart from the one that does the job :)
> Go into Edit (red) mode. Select the steps. Press (not hold) Instrument button. Rotate the jog wheel.

Yep I get that, but if you only select one step, it does work by holding instrument and turning the jogwheel. I assumed it would work the same way with multiple steps. Also, you can't see what the sample is if you are not holding instrument. You have to check first and then record. It would be handy if you could do it for multiple steps while viewing the instrument pool using the jogwheel. Thanks.