polyend/TrackerBetaTesting #185: [1.1.20] Exported or recorded song, slowly decreases tempo out of sync

MegaTurrican, 2020-10-08:
I exported a song (26 patterns) at 172 bpm, inserted it in my DAW and its tempo was increasingly lagging and becoming out of sync. After that, I recorded the same song live from the Tracker and the same issue occurred. I know this issue has been reported and closed, but unfortunately it still seems present.

Jacknametrouble, 2020-10-08:
Were you using any tempo Fx's in the project?

MegaTurrican, 2020-10-08:
No, the tempo was fixed at 172 bpm.

jaap3, 2020-10-08:
I can confirm this happens for me as well on 1.1.21.

First I exported stems with the Tracker in full control over clock and transport. Then I synced the Tracker with Reaper as clock/transport master over MIDI jack and recorded the Tracker's output.

I then imported the stems in the same Reaper project. All should be at 180 BPM, however I can clearly see that the exported stems are getting more and more out of sync during the duration of the track.

I have to say that the stem export function has much improved the last couple of firmware updates. The tracks used to be way more out of sync, even drifting individually. Now they seem to be mostly in time with each other. Hopefully it can be improved even further so the BPM stays stable over time.

loukltd, 2020-10-08:
I have this as well, tempo says 130 but when you export it's 129.13, happened on 30 projects.

clickysteve, 2020-10-26:
Confirming that this is still happening in v.1.2.0.

Related: https://github.com/polyend/TrackerIssues/issues/281

@loukltd @MegaTurrican what are you using to detect the actual bpm of the exported stems?

MegaTurrican, 2020-10-26:
@clickysteve I am using FL Studio's beat detect feature and have also confirmed the detected tempo in Reaper.

ambv, 2020-11-16:
We'll be solving this as part of #281. Marking this as duplicate even though this one was first. Thanks for your report, @MegaTurrican.

closed 2020-11-16 by ambv