polyend/TrackerBetaTesting #186: Suggestion to lower the preview volume of the sample loader.  Or question of another way.

modmodmodmod, 2020-10-08:
I'm grateful to the Polyend for the great updates you have made.  Thank you so much.

But with the latest 1.1.20 update, I find it difficult to balance between the playback sound of pattern and preview sound of sample loader.

 I use the methods that are often exchanged in common mixdowns these days as a reference, and set the volume in the following order.

 1, Set the Master volume fader to 0db with Master 1/3.

 2, Set all volume faders to 0db with Master 2/3.

 3, Adjusts the gain of the instruments so that the meter of each track balances the overall volume while ensuring the necessary headroom with the effect as premix data in Master 2/3.

 4, While raising volume of monitor speaker, lower the Master volume fader of Master 1/3 and set it to the volume suitable for listening.  (It is around -20db in my environment.)

 However, if I preview the samples in sample loader while playing back pattern in this state, the samples will be previewed at a loud volume.

I can adjust the volume of sample preview with Master volume in Master 1/3, but lowering this will also lower the volume of pattern playback, so I'll have to increase the instrument gain.  However, as soon as I increase the gain of instruments, The clip occurs on Master 2/3.  And if I lower the fader with Master 2/3, the level of playback sound of the pattern will drop.

 I personally feel that the preview volume setting of the sample loader is too large, and I would like you to lower this setting, but as Polyend, what kind of volume balance adjustment is assumed as reference?

bambrose, 2020-10-08:
I'd agree with this. The sample loader preview volume is a few db too high. I'd suggest this is a bug in fact

bambrose, 2020-10-09:
This is a duplicate.. there is (now) another thread where this is raised as a bug

closed 2020-10-09 by andrzejmagierski2