polyend/TrackerBetaTesting #189: {1.1.21] Freeze - External Midi clock from pattern view

rpc9943, 2020-10-08:
## Hi, sorry you had trouble!

### Steps to reproduce

1. enter 3 slots into song view
2. change slot 3 to pattern 2
3. press play from external transport

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
often

* Does this bug happen in previous firmware versions?
no

* Does this bug happen in every project?
yes so far

Jacknametrouble, 2020-10-09:
Sending MIDI clock/transport/notes over USB (Live 10) to the Tracker while in the Song mode is generally freezing it each time (need to unplug the USB cable to bring it back).

jaap3, 2020-10-09:
I can confirm my unit freezes as well on 1.1.12 with Reaper as a clock/transport master when using MIDI over USB. With MIDI over jack it works just fine.

stevesomeone, 2020-10-15:
It is not only midi over USB, the bug is present over midi jack as well.
https://github.com/polyend/TrackerIssues/issues/199#issue-718735066

closed 2020-10-16 by piotr3500