polyend/TrackerBetaTesting #190: [1.1.21] The preview volume of sample loader is a little too loud for the sound of pattern playback.

modmodmodmod, 2020-10-09:
I posted this issue as a feature request for # 186, but after that I reconsidered that it was as influential as a bug. I also received a message with the same opinion. So I reposted this issue as a bug report. Sorry for your inconvenience, but could you consider adjusting this issue and delete the post that was judged to be inappropriate?

* How often does this bug happen? (every time)

* Does this bug happen in previous firmware versions?（It also happens in 1.1.20）

* Does this bug happen in every project?（yes）

closed 2020-10-09 by andrzejmagierski2

MegaTurrican, 2020-10-13:
@andrzejmagierski2 @Jacknametrouble This bug is still present in 1.1.22 The preview volume of the samples is still high compared to the pattern; it should go back to the preview volume on previous firmwares before the dB introduction - or it should be proportional to the master volume. If you have a quiet track that you need to have the master volume all the way up, the sample preview will kill your speakers in the latest firmwares with dB values.

modmodmodmod, 2020-10-13:
I agree.  Polyend once fixed this problem, and it was a little better than before, so I couldn't say it.  Thank you.

Jacknametrouble, 2020-10-14:
Notice that the volume of samples recorded from any external source or FM radio is always on the exact same level as the original source (plus Gain). But their payback volume depends on factors as the volume set in Instrument Parameters and the track volume from Master section mixer.

geoffreyclaret, 2020-12-12:
fixed? Preview volume is still too lound in 1.2.2 and 1.3 betas.. Is the worst issue for me at the moment..

trutypesounds, 2021-09-26:
this is still in 1.4.0!
@ambv

NoMoreFreezesPls, 2021-10-14:
ITS SO LOUD!!!!!!!!!!!!!