polyend/TrackerBetaTesting #192: [1.1.21] Entering Master page jumps directly to page 2/3

Intothegroove, 2020-10-09:
When pressing the Master button, it does a slight glitch and jumps directly to page 2/3 instead of the first page 1/3

* How often does this bug happen? Every time

* Does this bug happen in previous firmware versions? I noticed it happened in 1.1.20

* Does this bug happen in every project? Yes

stevesomeone, 2020-10-09:
I have noticed this as well. It is also accompanied by red level meter glitches when you initially enter Master mode.

Jacknametrouble, 2020-10-09:
Confirming, after toggling the unit on and using the Master key it works properly and turns on the 1/3 screen. But after getting to 2/3 screen and toggling between other screens it always jumps back to 2/3. No matter if you'll set it to 1/3 or 3/3.

bambrose, 2020-10-09:
ah yeah.. I had wondered why I was hitting master more than normal, confirmed for me as well

closed 2020-10-12 by andrzejmagierski2