polyend/TrackerBetaTesting #196: [v1.1.20] Most steps inside a single track sound at lower volume (-6dB or so) [until more tracks are playing simultaneously]

Sft-mlk, 2020-10-09:
**_Explanation_**; _[In this case Kick & Snare shares the same track]_

All sounds good till;

**1.** _I **mute the rest of the tracks**_, then automatically the kicks and snares low their volumes. _(and I'm sure that I'm not confusing the energy of the master with the energy of a single track)_

**2.** **If that track is the only track with any information of the pattern**. _(Other tracks are empty, wich is similar to 'mute' function)_

**3.** Curiously, It becomes **more evident if the master volume is lower.** At 0dB is more concealed. In fact I compared lowing the 'master' of my portable speaker instead of the Tracker master, and that effect is not present. So is not a psychoacoustic effect.

--

So;

I notice it clearly specially with short samples (kicks, snares...), not that clearly with longer samples. 

I actually can hear better the drums if the rest of the tracks are playing too.

Intothegroove, 2020-10-09:
I am noticing this too, I have a project with all 8 tracks used and on two of them I have a pad playing. Whenever I press play and start the pattern, especially if I enter the pattern window from another menu, the pads are so much more louder than the drums and the volume resets to it's proper volume after the pattern starts over.

Sft-mlk, 2020-10-09:
Yes, I think is another issue, maybe related in some way with this one, because the sensation of energy loss is pretty similar. The first hit of a pattern sometimes loose its volume. If that first sound is a kick, snare, hihat... Then the sensation is similar to loosing the quick attack time of that first step.

piotr3500, 2020-10-10:
Please check on 1.1.22 and let us know if it the problem still exists.

Sft-mlk, 2020-10-11:
[v1.1.22] Still present, same behavior described.

piotr3500, 2020-10-11:
I have to ask for the video...

Sft-mlk, 2020-10-11:
Not perfect quality but it's perfectly audible, **check drums before and after I mute the tracks 3** (bass) **and 4** (piano samples) [https://youtu.be/UIedJggoEZ0](url)

Sft-mlk, 2020-10-11:
I can't open the video just clicking on it, I had to copy the link and paste it on my navigator.

I dont know if that is normal on github or what. Sorry if it's my fault.

piotr3500, 2020-10-11:
Hard to judge from the video. 
The volumes might differ after muting certain track because of the limiter.
Please try setting the threshold to the max value.

Sft-mlk, 2020-10-11:
> Please try setting the threshold to the max value.

Done, still happening. I also tried with a longer attack for the limiter and same.

piotr3500, 2020-10-11:
Then perhaps having your project would help.

Sft-mlk, 2020-10-12:
_I have good news I think 🤔 this morning finally I found a common point,_ **I notice that it only happens with [this project + JBL portable speaker].** _**So weird** cause is_ **not happening with other projects + JBL speaker**, _even with the same drum samples._ 

_**BUT the good new is I was able to record the project to the computer track by track without having this issue, is just a very weird listening issue**._

_Anyway I let the link of the project here if you want to check the weird thing._ https://mega.nz/folder/n2wiRbga#44eoYQsjIitRKH9j_LGEmw

piotr3500, 2020-10-12:
No worries. Thanks for helping us make the firmware as good as it can get.

closed 2020-10-12 by piotr3500