polyend/TrackerBetaTesting #199: 1.1.21 Tracker freeze in Song mode

stevesomeone, 2020-10-11:
Testing the new Song mode implementation. (red and green modes).
Tracker freezes totally
1. Enter song mode
2. shift between green and red modes.
3. on the third shift, the tracker sequencer stopped and froze. no response from any button and will not start.
4. requires reboot.
* How often does this bug happen? I have reproduced this twice in 3 tests.

* Does this bug happen in previous firmware versions? No

* Does this bug happen in every project? Yes

piotr3500, 2020-10-11:
I can't reproduce it. A video would give us more hints.

ambv, 2020-10-11:
I also cannot reproduce it.

Jacknametrouble, 2020-10-11:
Are you using external MIDI clock over USB maybe?

stevesomeone, 2020-10-11:
I have updated with links to 2 videos.
After more investigation, this appears to be an issue with external midi clock/transport.
You can see both; the inability to record info; and the random recording onto different tracks.
In a second video, you can see that you cannot step back to step rec mode.

https://www.dropbox.com/s/rv9q54wxrabivuf/Live%20rec%20bug%201.mp4?dl=0
https://www.dropbox.com/s/xpngjy0py0outna/Live%20rec%20bug%202.mp4?dl=0

ambv, 2020-10-12:
I can reproduce this.

piotr3500, 2020-10-16:
fixed in 1.1.23

closed 2020-10-23 by shuler323