polyend/TrackerBetaTesting #203: 1.1.22 Broken samples in imported mod files (worked before)

LarsForsberg, 2020-10-11:
## Hi, sorry you had trouble!

### Steps to reproduce

When importing mod files in version 1.1.22, the samples are broken and distorted. This worked fine in version 1.1.17, but seems to got broken in 1.1.19.

### Attachments and links

Test importing this mod file in version 1.1.17 and then in 1.1.22. The samples are distorted in 1.1.22 and cannot be used:

https://www.dropbox.com/s/1o0vi1gt21xfrxp/o3.mod?dl=0

### Other important information

* How often does this bug happen? Every time.

* Does this bug happen in previous firmware versions? Not in version 1.1.17. 

* Does this bug happen in every project? I have tried two mod-files so far, both get distorted samples.

If I import a mod-file in version 1.1.17 and save it as a project, I can load the project in version 1.1.22 and it sounds fine. The sample bug seems to be only related to importing mod-files.

Jacknametrouble, 2020-10-11:
We'll improve importing MOD files in the future. Thanks for reporting that.

LarsForsberg, 2020-10-12:
Thanks. The MOD import feature was already very useful in v1.1.17, but not in v1.1.22 because of broken samples. If possible, can you revert back the import code to v1.1.17, which worked fine in regards of the samples? I can upload a video later demonstrating what I mean.

Jacknametrouble, 2020-10-12:
We'll take a closer look at what's going on. Thank you for the hint.

shuler323, 2020-10-22:
@LarsForsberg  1.1.23 (and above) should work fine, i've tested this  .mod on 1.1.25 and it sounds nice, please reopen issue if bug still occours

closed 2020-10-22 by shuler323

LarsForsberg, 2020-10-24:
Works great! Thanks a lot!