polyend/TrackerBetaTesting #206: [1.1.0] Visualize Instrument Modulation

sebsonf, 2020-10-12:
In the instrument modulation mode (window 1/1) it would be nice to see the actual modulated parameter value together with the bias / initial value of modulation that is set. This could potentially be a colored horizontal line above (in the z plane) the bias column. Potentially one line per instrument voice.

Jacknametrouble, 2020-10-12:
Dear, there's a published v1.1.22 firmware available for download here at the Github. We're currently working on v1.1.23.
Please download and update your unit, read the changelogs, then start posting. Thank you,

sebsonf, 2020-10-12:
Hi sorry, I apologize for that. I was going through all the closed and open issues, but I seem to have missed the changes from before GitHub. I have now updated to version 1.1.22. Thanks for your great work to the entire team and for being so open and responsive for feature requests!

closed 2020-10-12 by Jacknametrouble