polyend/TrackerBetaTesting #208: 1.1.22 File Transfer Mode not read in computer

muhfugga, 2020-10-13:
Hello, I'm using firmware 1.1.22 and am not sure if I'm doing something wrong with the file transfer mode. I connect to my computer and when I put the tracker in file transfer mode and the tracker does not show up on my computer. The tracker just says file transfer mode on it and nothing else happens so I just press end. Not sure if I'm missing something simple, sorry.

Just-Peter, 2020-10-13:
I can confirm that 'storage mode' is working fine. After enabling 'file transfer mode' the Tracker shows up in my Explorer on Windows 10.

jaap3, 2020-10-13:
On MacOS I had to use [Android File Transfer](https://www.android.com/filetransfer/) to access the Tracker in file transfer mode. It seemed to work fine.

LarsForsberg, 2020-10-13:
Another alternative on MacOS is to mount it using simple-mtpfs. It can be installed using brew. It is very simple. I just write:

simple-mtpfs -f polyend/

on the command line. I can confirm it is working fine.

muhfugga, 2020-10-13:
Hey, thanks for this. I downloaded it and it works with the tracker. Thank you.

closed 2020-10-13 by muhfugga

ambv, 2020-10-13:
I will add this as a FAQ entry.

otternet, 2020-10-18:
Hi - I downloaded Android File Transfer, SyncMate and also Commander One for the MTP driver for Mac. It still doesn't mount and File Transfer Mode doesn't seem to work. I am using MacOs Catalina 10.15.6

Jacknametrouble, 2020-10-18:
Are you running the newest firmware? Is the file transfer mode turned on in the Config menu?

otternet, 2020-10-18:
> Are you running the newest firmware? Is the file transfer mode turned on in the Config menu?

Yes I am. I’m on v1.1.23 now. I’ve turned on FTM, tried launching the apps first and then then on FTM and vice versa, but to no avail. Are you mounting it on MacOS too?

Jacknametrouble, 2020-10-18:
Yes, no problems with Mojave.

otternet, 2020-10-18:
Direct from Tracker to your Mac? I am using a USB hub to connect as my Tracker is the “square” button version and from what I understand it will not work unless it’s a USB C to A cable as supplied. I tried using C to C direct to my MacBook and the Tracker won’t even power up.

Jacknametrouble, 2020-10-18:
The cable needs to be USB C to A, have you tried another one? Please try without the USB hub, just a straight connection.

otternet, 2020-10-18:
I’m using a MacBook and it’s all C 🤦🏻‍♂️

The cable I am using is supplied from Polyend. It works fine with powerbank or direct outlet. I’m guessing it’s the hub then as it’s powering the Tracker and also as the cable for transfer.

Jacknametrouble, 2020-10-18:
This may be a good guess.

IOSAudio, 2020-11-17:
Here on OSX I get nothing mounted, on WIN10 I see the tracker and card, it has the used space correct but I see no files it says it is empty:

Full Speed device @ 24 (0x00270000): .............................................   Composite device: "Tracker"
    Port Information:   0x0018
    Number Of Endpoints (includes EP0):   
        Total Endpoints for Configuration 1 (current):   11
    Device Descriptor   
        Descriptor Version Number:   0x0110
        Device Class:   0   (Composite)
        Device Subclass:   0
        Device Protocol:   0
        Device MaxPacketSize:   64
        Device VendorID/ProductID:   0x16D0/0x0E9F   (unknown vendor)
        Device Version Number:   0x0277
        Number of Configurations:   1
        Manufacturer String:   1 "Polyend"
        Product String:   2 "Tracker"
        Serial Number String:   3 "4294967295"
    Configuration Descriptor (current config)   
        Length (and contents):   211
        Number of Interfaces:   5
        Configuration Value:   1
        Attributes:   0xC0 (self-powered)
        MaxPower:   100 mA
        Interface Association   Communications-Control
            First Interface   0
            Interface Count   2
            Function Class   2   (Communications-Control)
            Function Subclass   2
            Interface Protocol   1
            Function String   4 "MTP"
        Interface #0 - Communications-Control   
            Alternate Setting   0
            Number of Endpoints   1
            Interface Class:   2   (Communications-Control)
            Interface Subclass;   2
            Interface Protocol:   1
            Comm Class Header Functional Descriptor   
            Comm Class Call Management Functional Descriptor   
            Comm Class Abstract Control Management Functional Descriptor   
            Comm Class Union Functional Descriptor   
            Endpoint 0x87 - Interrupt Input   
        Interface #1 - Communications-Data/Unknown Comm Class Model   
            Alternate Setting   0
            Number of Endpoints   2
            Interface Class:   10   (Communications-Data)
            Interface Subclass;   0   (Unknown Comm Class Model)
            Interface Protocol:   0
            Endpoint 0x04 - Bulk Output   
            Endpoint 0x83 - Bulk Input   
        Interface #2 - Audio/Streaming   
            Alternate Setting   0
            Number of Endpoints   2
            Interface Class:   1   (Audio)
            Interface Subclass;   3   (Streaming)
            Interface Protocol:   0
            Uknown Interface SubClass Type   
            Uknown Interface SubClass Type   
            Uknown Interface SubClass Type   
            Uknown Interface SubClass Type   
            Uknown Interface SubClass Type   
            Endpoint 0x06 - Bulk Output   
            Class-Specific AS Audio EndPoint   
            Endpoint 0x85 - Bulk Input   
            Class-Specific AS Audio EndPoint   
        Interface #3 - HID   
            Alternate Setting   0
            Number of Endpoints   2
            Interface Class:   3   (HID)
            Interface Subclass;   0
            Interface Protocol:   0
            HID Descriptor   
            Endpoint 0x81 - Interrupt Input   
            Endpoint 0x02 - Interrupt Output   
        Interface #4 - Image ..............................................   "MTP"
            Alternate Setting   0
            Number of Endpoints   3
            Interface Class:   6   (Image)
            Interface Subclass;   1
            Interface Protocol:   1
            Endpoint 0x88 - Bulk Input   
            Endpoint 0x09 - Bulk Output   
            Endpoint 0x8A - Interrupt Input

bn-mk, 2021-01-04:
same issue using a hub as mac only has usb c ports :(