polyend/TrackerBetaTesting #214: MIDI Instrument Randomization

jdlinck, 2020-10-14:
Hi Polyend!!

Although, the Tracker is a sampler at its core, I would love to be able to use the Trackers FX to manipulate MIDI instruments/Channels the way it does with samples. This would be great fun when pairing eurorack sequencers to Tracker. I think you could randomize gates in Note Mode say with the Polyend Poly, but you wouldnt be able to send note info as well. This would allow you to use the eurorack sequencers CV out for pitch, "velocity", or as a modulation for each voice or channel. 

Please consider and if anyone knows how to achieve this another way, please let me know. 

Best

Just-Peter, 2020-10-14:
If I understood well, after the release of 1.2.0 the Polyend devs will focus on midi.

visualex, 2020-12-27:
One interesting feature could be to send random values at once to all 119 MIDI CC, possibly skipping some of these either randomly or by configuration? First 11 can probably be skipped.

The goal here is to have an instrument attached on some MIDI channel and randomize its current settings, leading to a "randomize function" for gear that does not have this feature. 

Many happy accidents create cool sounds, like on Elektron gear (randomize page), Argon8.