polyend/TrackerBetaTesting #215: Clock multiplier/divider

anemlife, 2020-10-14:
Sometimes I would like the PT to play at half the speed of the incoming clock, and sometimes twice the speed. 

To achieve this at the moment I use some modular devices (Squarp Instruments Hermod / ALM Pamela New Workout for example) and it works fine but that means I have to carry my modular with me when I want to change the clock behaviour. 

That would be great if the PT could manage the clock speed for MIDI output and inputs internally (1/4 1/8 1/16 1/32 for example).

sebsonf, 2020-10-14:
I think this parameter would make a cool performance effect as well

Stazma, 2020-10-14:
I've been asking this since the early beta, I think it's in the pipeline for later, at least I hope so.

ArtFluids, 2020-10-17:
Yeah I want to throw in another request for this. Especially with the awesome expand/shrink pattern it's a great way to pack more notes/FX into a pattern but keep the same tempo. You don't want to expand a pattern, double the tempo and have all your synced devices going double time.

djkurzman, 2020-11-12:
> Yeah I want to throw in another request for this. Especially with the awesome expand/shrink pattern it's a great way to pack more notes/FX into a pattern but keep the same tempo. You don't want to expand a pattern, double the tempo and have all your synced devices going double time.

I agree on the expand and shrink. I went ahead and made a separate feature request for that.
https://github.com/polyend/TrackerIssues/issues/353