polyend/TrackerBetaTesting #217: Beat Slice mode; Select a 'slice' automatically just touching its associated pad, not changing the function on the screen button.

LetsSeeIfWeSolveThis, 2020-10-14:
Would be nice if when you press a pad then the Tracker select automatically that slice, showing the yellow bar turning to red. Would save a lot of time.

Or maybe a switch button to turn on/off this possible new function, for other users who are get used to do in the current way.

HakkahFR, 2020-10-14:
100% same opinion , i actually don't use the beat slicer for this reason, ultra buggy

keith-macleod, 2020-10-16:
I actually take exception to the previous comment, the beat slicer works well. It's quite fast to navigate and edit with just the arrow keys and jog wheel.
However, I have to agree that from an interface standpoint, this just makes sense.
I'd love to see some things like threshold for autoslice, slice into equal segments, and snap to zero-crossing. Then it would be pretty feature-complete.

danapeks, 2020-10-19:
Another +1 for adding this option. On old Akai samplers you could press prog edit then hit key on your midi keyboard and it would take you to that edit page. SUPER useful and speedy.

otternet, 2020-10-20:
Add me to this too. So much faster rather than using the jog wheel to move along the slices.

paalweb, 2020-10-29:
+1 from me as well. I think it would make slicing faster.

vaeinoe, 2020-11-06:
+1 for this one, as mentioned in the other (closed) issue. Having a pad selection for active slice would speed up slicing workflow a LOT and make it fun. Now it's doable but just tedious enough that you don't want to start working on slicing a sample unless you're pretty certain the track calls for that particular one.

antistatikk, 2021-01-08:
I‘m all for this one too!

Nodoubtyall, 2021-02-20:
Yes, it would be much faster! 🤞

closed 2021-12-02 by Jacknametrouble