polyend/TrackerBetaTesting #218: [1.1.22 ] Position in Wavetable is not saved

doomglue, 2020-10-14:
If I load a wavetable with a set of different waveforms and choose position 2 or 3 or whatever, upon reboot of the tracker the position is always reset to position 1.

* How often does this bug happen? (every time, often, "it only happened once")
Seems to persists with my current project, but frankly I did not test it with a fresh project

* Does this bug happen in previous firmware versions?
I never noticed this before.

piotr3500, 2020-10-14:
We didn't have this issue since 0.9 ;)
Please observe what set of action can cause it.
Thanks!

doomglue, 2020-10-14:
I used the additive wavetable, and wanted Position 2, in my set it was
always reset to Position 1.

I just found out, when I save with Position 5 and reopen, the Position is
then Position 4.

doomglue, 2020-10-14:
[deadpan dock.zip](https://github.com/polyend/TrackerIssues/files/5381126/deadpan.dock.zip)

closed 2020-10-15 by andrzejmagierski2