polyend/TrackerBetaTesting #221: [1.1.22] Screen glitch in pattern mode

isbtn01, 2020-10-14:
The screen sometimes glitches when going from other modes to pattern mode. Everywhere seems to be normal. This happened just 2 or 3 times during a 2 day period and i dont really know whats causing it. 

Link to a vid here

https://www.dropbox.com/s/f2ciyadaj52l6k0/IMG_5243.MOV?dl=0

* How often does this bug happen? 
Happened 2-3 times

* Does this bug happen in previous 
No

* Does this bug happen in every project?
Happened in two different project
Thanks

piotr3500, 2020-10-14:
We didn't have this issue since 0.9 ;)
Please observe what set of action can cause it.
Thanks!

isbtn01, 2020-10-14:
Will keep an eye on that and let you know when i find out. Thanks a lot!

isbtn01, 2020-10-17:
Update: 
Updated to 1.1.23
Happened again only in pattern mode when switching from partial 8 channel view to normal pattern view. 
When reseting the machine it was gone and i couldnt replicate it. Heres a vid again:
https://www.dropbox.com/s/rcgippbr9jz7ql0/IMG_5280.MOV?dl=0
Will be trying to replicate it.
Thanks!
Iso

ambv, 2020-11-16:
We will be fixing this issue as part of #325. Thanks for your report, @isbtn01.

closed 2020-11-16 by ambv