polyend/TrackerBetaTesting #235: [1.10 & 1.22] Note entry stops working / becomes disjointed

math0ne, 2020-10-16:
#148 

# Steps to reproduce

Hard to reproduce but I have experienced this on all the firmware versions I have tried.  Saving and reloading the song seems to fix this *sometimes*.

1) Work on a song until it becomes complex
2) Switch to a new pattern
3) Use midi in for note entry (seems to happen more but I have had this happen without a midi controller as well)
4) Note entry doesn't work
5) Enter note using polyend
6) Note entry works
7) Switch to a new pattern
8) Switch back and pattern was not saved

### Attachments and links

https://gph.is/g/a9Y2k38

### Other important information

* How often does this bug happen? 

Prety much every song i work on this happens eventually.

* Does this bug happen in previous firmware versions?

Yes on all firmware versions I have tried.

* Does this bug happen in every project?

Doesn't seem to happen until a project gets complex.

piotr3500, 2020-10-17:
1.2.23 fixes it.
Please install and check.

closed 2020-10-17 by piotr3500

math0ne, 2020-10-19:
Confirmed, thanks!