polyend/TrackerBetaTesting #236: 1.1.23 Live Record Mode issues still

stevesomeone, 2020-10-17:
This one is a persistent problem.
The video is self explanatory as I have included text.
The issue is as previously reported in various firmwares, but there is now an additional problem with intermittent glitch of the display/scrolling. (Pattern Mode)

[https://www.dropbox.com/s/rfl8zh3aj2jj89p/1.1.23%20Live%20rec%20mode%20issues.mp4?dl=0](url)

* How often does this bug happen? (every time, often, "it only happened once")  Every time

* Does this bug happen in previous firmware versions? Yes, with the exception of the screen glitch which is new to 1.1.23

* Does this bug happen in every project? Yes

sakke-makke, 2020-10-20:
My issue is almost as bad, or I would say even worse. I'm trying to record live, but every third or 4th step goes into a different track! I've painted only the track that I want to record on, but it makes no point of live recording, if every once in a while notes are flying to the wrong track.

Jacknametrouble, 2020-10-20:
@sakke-makke please update to 1.1.25 and check if the issue is solved.

MirEko676, 2020-10-21:
@sakke-makke  I also have the same issue with recording jumping lanes. I will check 1.1.25

shuler323, 2021-01-05:
file under link deleted, please open new issue if problem occurs in v 1.3.0 or above

closed 2021-01-05 by shuler323