polyend/TrackerBetaTesting #246: Slicer and fx

dramaone, 2020-10-17:
Sorry if this has been requested however I could not find this specific.

Having shopped a beat in slicer and using the sliced beat it would be nice to have the option to change each slice's pitch when programming in the pattern. I can obviously use the micro tune fx but would like to be able to have a semi tone fx. 
Without this I am forced to use samples like I would on Octamed rather than beat slice.

I just love to hit a snare and then hit another snare an octave lower. 

Many thanks

danapeks, 2020-10-20:
FYI - you can already do this, make sure you use Slice mode in Sample Playback rather than Beat Slice mode, then just change the note of the slice in the pattern screen.

dramaone, 2020-10-20:
> 
> 
> FYI - you can already do this, make sure you use Slice mode in Sample Playback rather than Beat Slice mode, then just change the note of the slice in the pattern screen.

Sorry I meant in Beatslicer mode

danapeks, 2020-10-20:
Ah ok, whats the difference? I've not been able to work it out any major differences other than the pitch slice thing

dramaone, 2020-10-21:
I believe slice mode doesnt spread the slices across each of your white buttons and so will only put the the highligted slice across all buttons. So if you highlight a snare within the sliced breakbeat, only the snare will be mapped across all buttons at different pitches

dirtysampling, 2021-07-05:
I think that dramaone has a good point. But in beat slice you can alerady pitch slices individually by adding the tune FX (U) and add/remove semitones. The point would be to have this function to tune the slices using the Note Lane in Slice mode (where you have to use one FX lane to call the slices).