polyend/TrackerBetaTesting #254: Slicer play mode - Thru

danapeks, 2020-10-19:
My feature request is related to Slice mode: It would be super useful if there was an option for it to play "Thru" the whole of the sample from where you have triggered a slice. So rather than just cutting off at the end of the slice it plays through the rest of the e.g. breakbeat from the start position of your slice.

This works rather like the 9XX Offset commands on old school trackers, as implemented as PXX on the Polyend Tracker, however the Position effect is only 1-100 so often not accurate enough to catch transients, whereas the slicer could do this perfectly. 

Ableton have a Thru mode on their Simpler device if you know that: (2:08 in this demo https://www.youtube.com/watch?v=o-mGmyW6-HY), really great for drum chopping/playback.

VinceKabel, 2020-10-20:
Yes! Great idea. I was thinking about this the other day after making the switch from using Ableton and simpler. Hopefully a straightforward thing to implement?

MegaTurrican, 2020-10-21:
Cool idea!

gherat, 2020-12-05:
yes please! I requested this too, I don’t understand why they made the slice stop at the beginning of the next slice. It very limiting musically.

autolog-statbot, 2022-01-03:
+1 for this to be added asap