polyend/TrackerBetaTesting #255: [1.1.23] System Crash in sample loader

danapeks, 2020-10-19:
Hi, awesome product thank you. 

First bug I have found is in Sample Loader screen. I am using the 16gb card it came with, and I go to load some samples which is fine but when I go back into the previous folder the system crashes, and I can see samples which I couldn't previously see as available, however I can't select any as the system has hung. 

This is happening everytime.with this specific folder and on different projects, but only seems to be with certain folders. Am not sure what the consistent link is.

Video link: https://streamable.com/zmdaf2

ambv, 2020-10-19:
If your `Think/` folder isn't too big, could you kindly share it with us? Pretty sure there's a sample in there that's not handled right.

danapeks, 2020-10-20:
Hey, thanks for getting back to me - just opened the folder and realised there are several potential issue files here which could be causing the crash!
At the root level there was a zip file, a rx2 file and some .asd files in the top level of it, which ableton automatically creates when you have used it in a project. Do you think any of those could cause the issue?
Be great if the polyend could 'overlook' the files that are not relevant, as the .asd files especially are all over my sample folders!

ambv, 2020-10-20:
We really can't tell without looking at your folder. It's probably either a single filename that's confusing the Tracker or a file that says is a WAV but isn't. If you could share this somewhere we'll be able to fix it quick.

danapeks, 2020-10-20:
Hi ambv, yes no probs - attached here: https://we.tl/t-eYApgnsuqA

wjakobczyk, 2020-10-23:
This is caused by ".wav" extension in one of the folders. A fix is coming, in the meantime you can just rename that.

closed 2020-10-23 by wjakobczyk

danapeks, 2020-11-01:
Hi there, I've upgraded to 1.2.1 which in theory fixes this issue but the same folder crashes doing the same thing, I wonder if there's another issue other than the .wav extension?

wjakobczyk, 2020-11-02:
Could you share the folder?

danapeks, 2020-11-10:
Hiya, sorry for the delay - here is a short video of the issue: https://streamable.com/opaq4h

I go to the Think folder and can see subfolders but no wavs in the Think folder, click in a subfolder then back and now the wavs show up.

When i try and go in folders with just wavs (no subfolders) the wavs don't show up. It's very weird. This is happening on 1.2.1 

I have attached the whole folder from my sample collection here (warning it is over 1GB): [Breaks folder](https://we.tl/t-c32IcJrlbd)

Would be very grateful if this could be fixed as it's a shame to not see any of these files in Tracker as chopping up breaks is the main thing I like to do! I love the tracker BTW, so much fun!

Thanks :)

wjakobczyk, 2020-11-12:
Hi, thanks for the archive.
I have tested with this and everything works fine, I enter the Think folder and can see all subfolders and wavs. Could you try applying the update again, and make sure your Trackers starts with 1.2.1 (the version is displayed during startup). BTW you can also upgrade to latest 1.2.2 - there is another fix for the sample loader.
If the problem persists, please check if it helps if you rename the folder "Lyn Collins - Think Break [96kHz 32bit stereo -8 Pitch].wav" to "Lyn Collins - Think Break [96kHz 32bit stereo -8 Pitch]" (no .wav ending).

danapeks, 2020-11-14:
Hi, thanks for checking - I deleted and re-transfered the folder and update firmware - all working now!! Thanks :-)