polyend/TrackerBetaTesting #258: Global swing/shuffle/groove/quantize

sakke-makke, 2020-10-20:
Polyend tracker needs a GLOBAL SWING!

I would LOVE to use a global swing when using Polyend Tracker. I would like to compose some 'uk garage' style beats, but at the moment that is next to impossible. With micro-move you MIGHT able to pull it, and even the old tempo trick (for example each 2nd step is 125bpm and every other like 140bpm) can excel it, but I wouldn't want to waste fx lines for that kind of sorcery in this day and age, where each of us have used to add easily some 'global swing/shuffle' to the groove (Akai mpc style for example). I know many of the Tracker users won't be after this kind of update, especially more avantgarde/ambient/glitch -trackers, but I would find it VERY useful for my future projects.

So this is my request in short terms: Would it be possible to add some 'global swing/shuffle/groove' to master effects?

MegaTurrican, 2020-10-21:
@sakke-makke I second that, excellent request, very useful. I would be even greater if you could pull swing/shuffle/groove per track

smitherines, 2020-11-17:
For the love of god please

Jurin, 2022-09-05:
Would love to see something like this! Making UKG beats would be awesome and not having to waste an fx slot would be great.