polyend/TrackerBetaTesting #259: [1.1.22] Midi unit freeze

sebsonf, 2020-10-20:
Yesterday I managed to freeze my unit while sending midi commands with the following pattern:

Track 1:
- sends midi cc messages A & B on a per step basis

Track 2:
- sends midi program change every 4th step:

Here is a screenshot of the pattern
[image]

This was working nice and I was really impressed! I had the pattern run for a few minutes and then it suddenly froze. Had to pull the power plug.

I will try to recreate the issue today.

Jacknametrouble, 2020-10-20:
Please update to 1.1.25 and check if the issue persists.

sebsonf, 2020-10-20:
Was not able to reproduce the issue

closed 2020-10-20 by sebsonf