polyend/TrackerBetaTesting #261: Relative volume effect

keith-macleod, 2020-10-21:
The current implementation of the volume effect is absolute, in that changing the instrument volume has no effect on the volume of notes with a volume effect applied.

This makes it a very lengthy process to balance the levels of instruments when mixing when the instrument is used across a number of patterns, especially when it's used on tracks with other instruments.

The ability to have the effect act in a way that can be affected by the instrument volume would be really nice.

piotr3500, 2020-10-21:
We have addition Relative Volume FX on the list, so it will be implemented, sooner or later.

keith-macleod, 2020-10-21:
Thanks, Piotr. Great to hear!

MetabaronCZ, 2021-01-23:
I support this enhancement. Volume Fx should be relative to instrument volume.

static-noise-PT, 2021-09-09:
Please yes

antf4rm, 2021-09-23:
Yes please. This is a must have.

closed 2021-12-02 by Jacknametrouble

TheGrizard, 2022-09-01:
This is top of my list too! My way of thinking is the volume fx is pre fader to the track mixer - very hard to balance finished track otherwise. Thank you!