polyend/TrackerBetaTesting #262: Song white blocks and text

dramaone, 2020-10-21:
Just finished a track and I feel it would really help if you could click on a white block in song mode and name it. 

Even if limited to four or five characters, you could call them for eg bass, drum1, vox1, fx1  etc and
then its clear what they are and you dont have to preview the audio. Dead good for a quick copy and paste

Thanks

jaap3, 2020-12-27:
Track renaming has been added in firmware version 1.2.0 (I think). It's available on the "Track Mixer 2/3" screen (press the Master button twice). After hitting the red [Rec] button the word "Edit" (in red) will appear in place of Mute. Hitting the button underneath the track will open up a menu that allows you to change that track's name.

dramaone, 2020-12-28:
and the text appears in the white blocks?

jaap3, 2020-12-28:
Assuming you mean these white block, then yes:

[image]
[image]
[image]

dramaone, 2020-12-30:
[image]

And these white blocks get renamed?

jaap3, 2020-12-30:
Ah, yes that isn't supported yet. I think #421 is a duplicate of this one then.

dramaone, 2020-12-30:
> 
> 
> Ah, yes that isn't supported yet. I think #421 is a duplicate of this one then.

i think its important, only naming tracks suggests you only use 8 samples/sounds. This tracker would never stay that static, its not a mixing desk where you have synths,samplers plugged into a channel. Its multiple patterns examples, sounds crossing over tracks.  The white blocks need to be identified, through text or colour  ( with shades )atleast.  Its key

dramaone, 2021-09-01:
please, i need this, i change drum patterns to frequently that i quickly lose track and melt in a sea of non identifiable white blocks. Imagine a daw that didnt label its audio clips!

TwistedGenetics, 2021-09-01:
I would love this as well

dias236, 2021-09-02:
It would be already super dope to have a small number identifying the pattern on the rectangle

dias236, 2021-09-02:
It would be Nice and Easier to use the song mode with color coded tracks, where patterns have distinct tones of their track color.