polyend/TrackerBetaTesting #263: Set step 1

HakkahFR, 2020-10-22:
A function to place the first step of a pattern using a single key. Which would avoid many manipulations of copy paste and use of insert/backspace

keith-macleod, 2020-10-24:
I think a pattern or track, or even selection rotate feature would mostly cover this case, except for the single key part. It would be a generally powerful feature for tweaking happy randomization accidents that don't line up on the beat.

PEtajaja, 2020-11-10:
I'd really love a rotate selection feature.. Maybe if you have something selected and you press insert, it rotates it rather than the current insert behaviour we have where anything that falls off the bottom of pattern is lost rather than rotating back to the top.