polyend/TrackerBetaTesting #264: Simple Improved Project Management

clickysteve, 2020-10-22:
### Allow Projects to be organised into subfolders

Currently it isn't possible to browse subfolders when loading a Project file. In other words, projects are only recognised by the Tracker under File > Open if they exist directly under the root 'Projects' folder on the SD card. If you place them into subfolders, the list appears blank. This means that over time it can be unwieldy to navigate through multiple projects. 

**Request:** Allow projects to be organised into subfolders which could then be navigated in a similar way to loading samples.
NB: I have searched and not found any similar open requests.

**Version tested on**: v1.1.25 Beta.

clickysteve, 2021-09-26:
Any update on this? It would solve the repeated problems folk have asking about the max 100 projects in a view cc: @ziej8asek

no-carrier, 2021-09-27:
Yes, this would be great!

Aisjam, 2021-09-29:
I would use this feature a lot, Managing sets and onstage organisation. Could see the same file browser used the the sample select screen been used to explore the project file.

Midisplitter, 2022-04-10:
Does anybody know if there is any update for this?

I've been searching all over Github and the facebook groups, but I can't find a single answer from anyone working at Polyend. It feels like they are just ignoring this "Feature", while it seems like something very basic.