polyend/TrackerBetaTesting #266: More midi-in options

Jampauljones, 2020-10-22:
I’ve checked out the other requests to make sure if this simple idea was already posted somewhere so excuse me if I didn’t see it.

In the midi config tab it would be nice to have a global pitch-bend and mod-wheel parameter for external midi controllers.

ROM72, 2020-10-22:
And assign these input to : for example filter, sample position ? It would be great. 
If velocity data could be assign to instrument parameters it would already be lovely <3

smitherines, 2020-11-17:
Mod Wheel / PB per midi channel - that would get around the cc limitations reasonably well

eholzbach, 2021-01-27:
I spent a little too long trying to figure out why an external controller plumbed through midi in on the tracker sent notes but wasn't passing pitch bend. Sounds like this isn't a thing yet.