polyend/TrackerBetaTesting #268: 1.2.0  Tracker freeze in Song Mode

stevesomeone, 2020-10-23:
When you enter song mode, then start the sequencer, the tracker freezes.
To reboot the tracker, you must pull out the usb power cord.
In most cases this occurs immediately. It will occasionally work as expected for a short time, but will freeze with stuck audio.

* How often does this bug happen? In all instances when external midi via jack is enabled.

* Does this bug happen in previous firmware versions? Yes it has been reported in previous firmwares.

* Does this bug happen in every project? Yes.

[https://www.dropbox.com/s/x21e7gefqmqfrs8/Tracker%20freeze%20song%20mode%201.2.0.mp4?dl=0]

rpc9943, 2020-10-23:
confirmed! it was same as it operated before! freeze bug that I had posted before

shuler323, 2020-10-28:
try 1.2.1  (same issue #199)

closed 2020-10-28 by shuler323