polyend/TrackerBetaTesting #269: Can others please advise if this is an error.

stevesomeone, 2020-10-23:
1. External clock and transport via midi jack enabled.
2. Set the tracker in Live rec mode.
3. Start the sequencer (via external)
4. Try to step back into step record mode.

When you enter live rec mode AFTER the sequencer has been started, there is no issue, as expected, you step back to step rec mode by pressing rec.
If you enter live rec mode before you start the sequencer, pressing rec will not get you back to step rec mode. 

The Tracker does respond and start as expected, but it is impossible to get to step rec mode.
It has been suggested that I should do it by pressing PLAY, but this is not correct, as pressing Play will stop the Tracker sequencer, regardless of what the transport or clock source is set to.

I cannot reason why my reporting this as an issue keeps getting closed without confirmation or fix.
If I am doing something wrong, or if this will never work out by entering Live rec mode before starting, then I will be happy to drop it and move on.
I simply need someone else to advise.

Thank you.

RadicalAns, 2020-10-23:
Which version are you running?

stevesomeone, 2020-10-23:
@RadicalAns Currently 1.2.0, but it first appeared in 1.1.23