polyend/TrackerBetaTesting #272: Crop samples too long to load - A feature for the Sample Loader Screen

teamoth, 2020-10-23:
Would it be possible to add a simple sample cropping feature in the Sample Loader Screen, for samples too long for your project?
This would be a great thing to have. Sometimes I want to add part of a longish sample and I don't have much space left in my project, or sometimes I'd like to use part of a really long recording that is over the maximum for a whole project.
Tim (or teamoth or timbient)

bambrose, 2020-10-23:
very cool idea

teamoth, 2020-10-23:
Or, maybe another approach... in the same way the device now auto-converts samples to the right format, could the device auto-slice samples that are too long, into segments that are samples of the maximum length? So that a long sample called 'pebbly.wav' would end up being: 'pebbly_1.wav', 'pebbly_2.wav', 'pebbly_3.wav'.
This way at least you could access the sample on your Tracker and chop bits out to save for use in other projects etc.
Tim (or teamoth or timbient)

Jacknametrouble, 2021-12-02:
done

closed 2021-12-02 by Jacknametrouble