polyend/TrackerBetaTesting #274: Low and highpass filters for delay and reverb

OnevenHub, 2020-10-23:
Would love more control over the reverb and delay. I think a simple high and lowpass filter would really help fine tuning the effects.

deixic, 2020-10-24:
I agree, and have also proposed this (closed issue #248).
Unfortunately, I think Polyend mistakenly thought of my request as per-send, instead of global.
I really hope they implement this, because without it the Reverb and Delay have limited use in a professional studio, and a Master EQ will not fix the problem of low-end buildup in the Reverb and Delay without also affecting the entire mix.
Here is a possible implementation idea:
(signal flow = instrument send > global Reverb HPF/LPF > Reverb > Master output):
[image]
[image]
https://www.youtube.com/watch?v=VhpwM8EoUPY

smt923, 2020-11-11:
hope this could be implemented, helps to mix delay/reverb and also gives more possibilites with the sound, would be really useful to  have this

closed 2021-12-02 by Jacknametrouble