polyend/TrackerBetaTesting #278: FX - Envelope Parameters

uija, 2020-10-24:
Hey!

I would love to be able to manipulate envelope parameters per step. For hihats modifying the decay and for sounds modifying the filter amount would be my number one targets. 
As there are too many options to provide a FX for them, maybe a modulation target list would be possible (like midi cc) just with more entries, as 5 might not be enough.

Jens

uija, 2020-10-24:
I am unable to add an enhancement label on my iPad :(

HakkahFR, 2020-10-31:
+1000 for this

60pfennig, 2020-12-22:
+10000 for this!

keith-macleod, 2021-02-27:
+1,000,000 for this. If there was a way modulate amplitude envelope decay, performance fx, step fx, whatever. It's mind-boggling how powerful this is for bringing a beat to life. All envelope parameters would be nice, but this one is just so crucial, along with gate time, but that's another topic.

x2db, 2021-04-14:
1Billion+  for this, but if there was a mod matrix, I would just take unused LFO (say fine pitch) and control the Decay with this LFO as I do on 10+ of my synths :)

closed 2021-12-02 by Jacknametrouble