polyend/TrackerBetaTesting #281: [1.2.0] Tempo of Export Song Stems not accurate

soundbylan, 2020-10-24:
When I use function Export Stems, then drag those files to my DAW, each of the tracks are slightly off tempo. The song start off on tempo, but as it progresses it increasingly gets off tempo.

In my example attached, I exported a song that is 70 BPM into my DAW (Reason) with tempo also set to 70 BPM. If you look at the ends, you see that each track ends at slightly different time, which is visual indicator of tempo being off. 

[image]
[image]

Here is link to video clip of song playing where you can hear in beginning it is on tempo with click track, but near end it gets off tempo with the click track https://youtu.be/nAIiks_G6bU (unlisted  / non-public youtube video). You can clearly hear the snares hitting at same time as click in beginning, but at end you hear the snares hitting after the click.

clickysteve, 2020-10-26:
I am experiencing the same issue. Exporting stems does not result in an accurate tempo. While the song may be consistent with itself, the tempo drifts out over time.

For example, song stems exported at 130BPM and placed into a Logic file at 130BPM, when the kick comes in on bar 4, it is at DIV 1, tick 49 past the click. 

[image]

By the time you go a few bars down the song (about 40 seconds in), the gap is now DIV 2 tick 21 past the click - so a clear drift.

[image]

While harder to measure, Pattern stems also export in an unexpected fashion: 

[image]

While more difficult to measure drift over shorter periods, they should not be exporting at different lengths.

doelman, 2020-11-20:
Even when you record all channels one by one (with the solo functionality) it's drifting. This makes it impossible to finish the projects I started on the Tracker

doelman, 2020-11-28:
Any estimates on when this is gonna be fixed? It's quite annoying not being able to use the stems of the Tracker in a daw, tried several times and with several songs but it's al a mess after exporting.

doelman, 2020-12-03:
This problem even got worse in the newest beta 1.3.0b2. Really curious what's causing this and if I could do something to avoid it.

shuler323, 2020-12-07:
Export function will be implemented in new way, you can expect fixing your issue in some new betas, which will be noted here.
(#415)

closed 2020-12-10 by shuler323

clickysteve, 2020-12-12:
:O

clickysteve, 2020-12-13:
@shuler323 when you mentioned that the method of export would change - can you provide any further details? I just downloaded the new beta firmware and attempted to export the song stems. It took quite a while, then appeared to be completed, but I can't find the wavs anywhere on the SD. I'm trying again, but perhaps this is an issue with myself rather than a different bug.

clickysteve, 2020-12-13:
Unfortunately this is still a problem with the new firmware: https://github.com/polyend/TrackerIssues/issues/512

headtopmusic, 2021-01-01:
I am struggling with this issue too. The length of stems appear inconsistent when exporting a track using firmware 1.3.0.