polyend/TrackerBetaTesting #283: [1.2.0] Screen glitch/midi issue?

thebradclay, 2020-10-24:
## Hi, sorry you had trouble!

### Steps to reproduce

Using track lane 8, with 64 steps. I added midi notes and 1) the notes played fine the first time through. On the second play through notes began being missed. On the 3rd play through no notes played. 2) I pressed the Rec button to follow the pattern and the screen glitched ( photo attached). I powered the system off. Unplugged the power and midi cable. Plugged back in power and midi and powered on. Couldn't reproduce the glitch but midi still isn't sending.

### Attachments and links

[image]

### Other important information

* Midi hasn't worked in this specific instance after 3 attempts. I will start a new project today and test. The screen glitch only happened this one time.

* This did not happen in the previous version

* Only happened in this project so far

Jacknametrouble, 2020-10-24:
Looks so pretty! Tracker autogenerated glitch art :) Nice catch thou!

ambv, 2020-11-16:
We will be solving this as part of #325. Marking as duplicate even though this issue was reported first. Thanks for the report, @thebradclay.

closed 2020-11-16 by ambv