polyend/TrackerBetaTesting #284: Move midi CCs out of global config

keith-macleod, 2020-10-24:
The current implementation of midi CCs has some severe limitations, the most troublesome for me right now being the fact that the 5 CCs that can be assigned are a global configuration. A very close second place is that only 5 can be accessed at any one time.

The problem as I see it is that each CC actually requires 2 parameters, which doesn't fit into the interface paradigm for track FX, so they've been limited to a small number to allow them to go in the track FX menu with the CC# baked into the effect, so the number is limited so as not to overwhelm the FX list with an extra 127 entries.

One solution would be to have a separate track type for MIDI tracks, but that's a whole paradigm shift that I doubt we're going to see.

So my most workable suggestion would be to have the CC config as a per-project setting, perhaps allowing for a few more. If they could be assigned per MIDI channel, that would be ideal, but I can see how that would add complexity that some wouldn't want.

jaap3, 2020-12-27:
With firmware 1.3.0 midi CCs are no longer configured globally instead the "Tracker can now be configured to send 6 distinct MIDI CCs per MIDI channel (configured per project in Instrument Parameters of MIDI instruments 51-67)".

keith-macleod, 2020-12-27:
@jaap3 Yes, haven't really had time to play with it yet, but it's an awesome update.

closed 2020-12-27 by keith-macleod