polyend/TrackerBetaTesting #287: Legato/tie track FX

keith-macleod, 2020-10-24:
Another in my long wishlist of midi-focused enhancements: A legato or tie effect which can slur the release of a note over the start of the next note, to enable portamento for external gear that works on that principle.

What would also enable this, and be quite useful generally, would be to have the ability to control step gate time, this would also cover my earlier feature request https://github.com/polyend/TrackerIssues/issues/138, which was only looking at it from a sample instrument perspective.

Any kind of implementation that avoids using one of the precious effect slots would be amazing, but even if it's an effect, it's still worth it to have.

gmli, 2021-01-01:
I was looking for a comment like yours ! Now with firmware 1.3.0 we can select only one track to record : Shift + Record + [arm or disarm any track]. Which is great, thanks Polyend !

I'm doing this : use my arturia keystep to live record on the Tracker on track 1 only (played over MIDI to my Minitaur).

My problem : if I press a note, then without releasing it I press another, the second note isn't recorded. It makes live recording in « mono track » practically impossible.

That would be great to have this possibility !

theorize999, 2021-05-16:
I like the idea of this being handled automatically, I'd really hate to waste an effect command telling it that the note is legato, I'd like to see a parameter we can set in the MIDI instrument screen that you could set and it will overlap the notes at the start of the new note unless you tell it not to with note off or note cut command (or the new gate command). It's clean and simple, and will only have to waste an fx command slot sometimes.

closed 2021-12-02 by Jacknametrouble