polyend/TrackerBetaTesting #293: [1.2.0] Midi Issues

thebradclay, 2020-10-25:
### Steps to reproduce

1) Create a midi pattern
2) Create a midi pattern in another track for the same instrument.
3) midi will play through one pass then stops.
4) Remove midi from one of the tracks and it sometimes works again.

Just recreated this using my Behringer TD-3.  And now midi is not working at all. Created a new file and nothing.

### Attachments and links

### Other important information

* Happens everytime

* NO

* So far yes

PEtajaja, 2020-10-26:
I think I already replied to this over on MW but I think it may be because you're getting constant overlapping notes.. The Tracker doesn't create a note-off until you tell it to. If you put a noteoff between each of your note messages where you don't want them sustaining, does it sort the issue?

shuler323, 2020-11-02:
1. is this happening over usb for example using ableton?
2. are you sure that there is no midi loop? note goes to TD-3, then it goes to Tracker, Tracker puts it in 'edit mode' to step, then playbacks this note, then note goes to TD-3, then it goes back to Tracker, Tracker puts it in 'edit mode' to step, then playbacks this note, then note goes to TD-3, then it goes back to Tracker, Tracker puts it in 'edit mode' to step, then playbacks this note, then note goes to TD-3, then it goes back to Tracker, Tracker puts it in 'edit mode' to step, then playbacks this note...

I can't reproduce it on FL or Ableton
:)

shuler323, 2020-11-12:
@thebradclay were my tips helpful? is this still happening in newer firmware?

shuler323, 2020-11-18:
@thebradclay, reopen this issue if bug still occurs with more info

closed 2020-11-18 by shuler323