polyend/TrackerBetaTesting #294: [1.2.0] Freezing except jog dial while editing pattern.

modmodmodmod, 2020-10-25:
I tried several times, such as restarting, removing the SD card, starting it, and then returning, but if it didn't change.
So I loaded SD card into the MAC and then put it back in the tracker and it recovered.
 It's now resolved, but I'm about to lose important data, so I'll report it.
Please check the video linked.

[https://drive.google.com/file/d/15hnoLxHMi2_4AB-mYsK--i8jwjUdEXTF/view?usp=drivesdk](https://drive.google.com/file/d/15hnoLxHMi2_4AB-mYsK--i8jwjUdEXTF/view?usp=drivesdk)

Jacknametrouble, 2020-10-26:
When trouble hits and issues persist, just remove the USB cable from its port and plug it back in. We'll investigate the issue. We may need a project where this has happened.

modmodmodmod, 2020-10-26:
Thank you.  It is possible to send a project immediately after freezing.  However, the problem was solved by reading SDcard into the Mac, so I was able to back up the data immediately after the solution.

takistap, 2020-10-28:
I experienced the same issue. Removing USB cable and SDcard didn't solve my issue. I had to flash the firmware and power on and off a couple of times.

piotr3500, 2021-01-08:
fixed in 1.3

closed 2021-01-08 by piotr3500